diydrones/APM_Log_File_Analyzer #206: Add Auto Command 19 - CMD_NAV_LOITER_TIME

bionicbone, 2016-01-09:
Add this to the auto analysis 

              <entry value="19" name="MAV_CMD_NAV_LOITER_TIME">
                    <description>Loiter around this MISSION for X seconds</description>
                    <param index="1">Seconds (decimal)</param>
                    <param index="2">Empty</param>
                    <param index="3">Radius around MISSION, in meters. If positive loiter clockwise, else counter-clockwise</param>
                    <param index="4">Desired yaw angle.</param>
                    <param index="5">Latitude</param>
                    <param index="6">Longitude</param>
                    <param index="7">Altitude</param>

rsilk1949, 2016-02-12:
Also needed for CMD 21 (Land).

bionicbone, 2016-02-13:
We currently monitor the following Auto Events, mainly because they are the ones that I have used:
Case 16   ' Navigate to New WayPoint
Case 20    ' Return to Launch
Case 22     'Take Off
Case 82     ' Navigate to New Spline WayPoint
Case 113    'CONDITION_CHANGE_ALT
Case 178    'DO_CHANGE_SPEED
Case 201    'DO_SET_ROI

Using SITL the available option in MP are:
correct parameters are not set here, **BOLD** are already completed
**CMD, 669918592, 29, 0, 16, 0, 0, 0, 0, -35.36326, 149.1652, 584.09 Home Set if available
CMD, 669918592, 29, 1, 16, 0, 0, 0, 0, 0, 0, 0 - New WP**
**CMD, 669918592, 29, 2, 82, 0, 0, 0, 0, 0, 0, 0 - New Spline WP**
CMD, 669918592, 29, 3, 18, 0, 0, 0, 0, 0, 0, 0 - Loiter Turns
CMD, 669918592, 29, 4, 19, 0, 0, 1, 0, 0, 0, 0 - Loiter Time
CMD, 669918592, 29, 5, 17, 0, 0, 1, 0, 0, 0, 0 - Loiter Unlim
**CMD, 669918592, 29, 6, 20, 0, 0, 0, 0, 0, 0, 0 - RTL**
CMD, 669918592, 29, 7, 21, 0, 0, 0, 0, 0, 0, 0 - Land
**CMD, 669918592, 29, 8, 22, 0, 0, 0, 0, 0, 0, 30.47 - Take OFF**
CMD, 669918592, 29, 9, 92, 0, 0, 0, 0, 0, 0, 0 - Guided Enable
CMD, 669918592, 29, 10, 222, 0, 0, 0, 0, 0, 0, 0 - Guided Limits
**CMD, 669918592, 29, 11, 201, 0, 0, 0, 0, 0, 0, 0 - Set ROI**
CMD, 669918592, 29, 12, 112, 0, 0, 0, 0, 0, 0, 0 - Delay
**CMD, 669918592, 29, 13, 113, 0, 0, 0, 0, 0, 0, 0 - Change Alt**
CMD, 669918592, 29, 14, 114, 0, 0, 0, 0, 0, 0, 0 - Delay to Distance from WP 
CMD, 669918592, 29, 15, 115, 0, 0, 0, 0, 0, 0, 0 - Yaw Angle
CMD, 669918592, 29, 16, 177, 0, 0, 0, 0, 0, 0, 0 - Jump
**CMD, 669918592, 29, 17, 178, 0, 0, 0, 0, 0, 0, 0 - Change Speed**
CMD, 669918592, 29, 18, 211, 0, 0, 0, 0, 0, 0, 0 - Gripper - **Not in Ardu Firmware enumeration file!!**
CMD, 669918592, 29, 19, 208, 0, 0, 0, 0, 0, 0, 0 - Parachute
CMD, 669918592, 29, 20, 206, 0, 0, 0, 0, 0, 0, 0 - Cam Trigger
CMD, 669918592, 29, 21, 181, 0, 0, 0, 0, 0, 0, 0 - Set Relay
CMD, 669918592, 29, 22, 182, 0, 0, 0, 0, 0, 0, 0 - Repeat Relay
CMD, 669918592, 29, 24, 183, 0, 0, 0, 0, 0, 0, 0 - Set Servo
CMD, 669918592, 29, 25, 184, 0, 0, 0, 0, 0, 0, 0 - Repeat Servo
CMD, 669918592, 29, 26, 202, 0, 0, 0, 0, 0, 0, 0 - DigiCam Configure
CMD, 669918592, 29, 27, 203, 0, 0, 0, 0, 1, 0, 0 - DigiCam Control
CMD, 669918592, 29, 28, 205, 0, 0, 0, 0, 0, 0, 0 - DigiCam Mount Control

bionicbone, 2016-02-15:
Also, we need to solve the following issues with auto command flight analysis:
Using SITL I have found further errors
- At the end of the flight we get "End of Saved AUTO Commands List"
- During the new mission at the top of the log command 0 should contain the current position of the Copter, so if there are 4 lines in the mission log there will be 5 lines in the new mission. Can we use this to set the starting position. What happens if GPS has not got a position?
- Because of the above alter the count from 1 to 4, ignoring zero.
- During the mission if Lon or Lng is zero the set the starting position to GPS, if that does have some good co-ordinates.