diydrones/APM_Log_File_Analyzer #233: Main Battery Failsafe Warning

rsilk1949, 2016-02-12:
Although I did not explicity set on the Battery failsafe, I did set the voltage level (FS_BATT_VOLTAGE) to 10.5. This will trigger a RTL even though FS_BATT_ENABLE is 0. I think it would help to show a warning in the failsafe settings section in this case.

============================================
FAILSAFE SETTINGS:
Main Battery = Not Activated. Will override if Battery voltage < [xx.x] volts.
============================================

bionicbone, 2016-02-12:
Log file for analysis is:
[2016-02-09 10-06-01.zip](https://github.com/diydrones/APM_Log_File_Analyzer/files/127758/2016-02-09.10-06-01.zip)

bionicbone, 2016-02-12:
Hi Robin,
I'm not sure this is true, the official documentation states:

Battery Failsafe Enable (ArduCopter:FS_BATT_ENABLE)

Controls whether failsafe will be invoked when battery voltage or current runs low

VALUE	MEANING
0	Disabled
1	Land
2	RTL
Failsafe battery voltage (ArduCopter:FS_BATT_VOLTAGE)

Battery voltage to trigger failsafe. Set to 0 to disable battery voltage failsafe. If the battery voltage drops below this voltage then the copter will RTL

Increment: 0.1
Units: Volts
Failsafe battery milliAmpHours (ArduCopter:FS_BATT_MAH)

Battery capacity remaining to trigger failsafe. Set to 0 to disable battery remaining failsafe. If the battery remaining drops below this level then the copter will RTL

Increment: 50
Units: mAh

Although not totally clear I think this means that it will if FS_BATT_ENABLE = 2, it would land if FS_BATT_ENABLE = 1 and be ignored if FS_BATT_ENABLE = 0

Sadly, in your log we only go below 10.5v for a split second, I can't remember what the trigger point is but I think its more than this:

[image]

I'll raise an issue for the document team to review the official documentation around this.

Kev

bionicbone, 2016-02-12:
Document Issue raised
https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/291
Kev

bionicbone, 2016-02-12:
Further reference to this documentation.

http://copter.ardupilot.com/wiki/failsafe-battery/

Low Battery warning (even with failsafe disabled)
Even if the battery failsafe is not enabled (i.e. FS_BATT_ENABLE = 0) the “Low Battery” voltage threshold and “Reserved MAH” numbers are used to issue a battery warning. This will result in:

“Low Battery!” message appearing on the ground stations HUD (if using telemetry)
Loud beeping if a buzzer is attached
Flashing LEDs
The only way to disable this warning completely is to set the “Low Battery” voltage and “Reserved MAH” values to zero.

bionicbone, 2016-02-13:
I've updated the issue diydrones/ardupilot-wiki-issue-tracker#291 for the documentation to be made more clear on the parameters page.

closed 2016-02-13 by bionicbone