diydrones/APM_Log_File_Analyzer #235: Super Simple Mode is On

rsilk1949, 2016-02-21:
Cannot see any reason to display this message.
None of the flight modes have the box checked in MP and cannot see any MP log entry to indicate this either.

bionicbone, 2016-02-22:
I'll check, what firmware version are you analysing? 

This is a bit flag within the DU32 codes that are written, the bit number has moved around over the versions and if honest when I did the coding for V3. 3 I did not check it the position was still the same.

In the DU32 there is all sorts, like if USB is connected, very important for AUVA-X2 hardware that has issues on previous firmwares and thinks it is powered via USB which causes some code to react differently.

Since you are seeing random changes on simple mode I'm sure it will be because they have changed the bit numbering of DU32, i'll need that firmware version to be able to review it.

For info, i dont support RC versions, (or DEV versions to some extent) because the code changes too quickly and often the logs change so the devs can see things happening then they take it away again. its never clear what will stay, or what the FMT will be for the released versions. Also, if you are good enough to be flying an RC version then you'll probably be more at home analysing the data in MP. Imagine that each version has a different flow in the code for each type of data line.

rsilk1949, 2016-02-22:
Ardu Version: V3.3.2 Build: (7f16e4d6)
    Ardu Type: APM:Copter
   FC Version: 0
 FC Frequency: 400Hz
   Frame Type: X
   No. Motors: 4

bionicbone, 2016-02-23:
Cheers buddy, I'll check the placement of this bit for this version.